aristanetworks/sonic #125: Optics not detected on 7808 on 202405.18

liamkearney-msft, 2025-04-04:
Hello,
We performed a EOS->SONiC upgrade on a staging device to SONiC 202405.18, and are seeing an issue where some optics are not detected / links are failing. These optics were previously detected and running fine on previous versions of sonic, and EOS.

The links in question are Ethernet96, 104, 112, 120 : 

```
   Ethernet88          48,49,50,51,52,53,54,55     400G   9100    N/A  Ethernet3/12/1   PortChannel9    down       up  QSFP-DD Double Density 8X Pluggable Transceiver         off                                                                                                         Ethernet96                      40,41,42,43     100G   9100     rs  Ethernet3/13/1  PortChannel13    down       up                           Unknown or unspecified         off                                                                                                        Ethernet104                      32,33,34,35     100G   9100     rs  Ethernet3/14/1  PortChannel13    down       up                           Unknown or unspecified         off                                                                                                        Ethernet112                      24,25,26,27     100G   9100     rs  Ethernet3/15/1  PortChannel13    down       up                           Unknown or unspecified         off                                                                                                        Ethernet120                      16,17,18,19     100G   9100     rs  Ethernet3/16/1  PortChannel13    down       up                           Unknown or unspecified         off  
```

```
liamkearney@***-lc03:~$ show int trans presence Ethernet112
Port         Presence
-----------  -----------
Ethernet112  Not present
liamkearney@***-lc03:~$ show int trans status Ethernet112
Ethernet112: Transceiver status info not applicable
liamkearney@***-lc03:~$ show int trans info Ethernet112
Ethernet112: SFP EEPROM Not detected
liamkearney@***-lc03:~$ show int trans eeprom Ethernet112
Ethernet112: SFP EEPROM Not detected
```

```
liamkearney@***-lc03:~$ show ver
                                                                                                                                                                                                                                                                                        SONiC Software Version: SONiC.20240532.18
SONiC OS Version: 12
Distribution: Debian 12.9
Kernel: 6.1.0-22-2-amd64
Build commit: 5f2bcda6fa
Build date: Tue Mar 11 10:21:23 UTC 2025
Built by: azureuser@863b3db2c000002

Platform: x86_64-arista_7800r3a_36dm2_lc
HwSKU: Arista-7800R3A-36DM2-D36
ASIC: broadcom
ASIC Count: 2
Serial Number: SGD21270368
Model Number: 7800R3A-36DM2-LC
Hardware Revision: 2a.00
Uptime: 00:46:04 up  7:28,  1 user,  load average: 2.43, 1.95, 1.88
Date: Fri 04 Apr 2025 00:46:04
```

```
sudo sfputil show eeprom
...
Ethernet112: SFP EEPROM detected
        Application Advertisement: N/A
        Connector: Optical Pigtail
        Encoding: 64B/66B
        Extended Identifier: Power Class 4 Module (3.5W max.), No CLEI code present in Page 02h, CDR present in TX, CDR present in RX
        Extended RateSelect Compliance: Rate Select Version 2
        Identifier: Unknown or unspecified
        Length(km): 2.0
        Nominal Bit Rate(100Mbs): 255
        Specification compliance:
                10/40G Ethernet Compliance Code: Extended
                Extended Specification Compliance: 100G PSM4 Parallel SMF
                Fibre Channel Link Length: Unknown
                Fibre Channel Speed: Unknown
                Fibre Channel Transmission Media: Unknown
                Fibre Channel Transmitter Technology: Unknown
                Gigabit Ethernet Compliant Codes: Unknown
                SAS/SATA Compliance Codes: Unknown
                SONET Compliance Codes: Unknown
        Vendor Date Code(YYYY-MM-DD Lot): 2019-04-04
        Vendor Name: LUXTERA
        Vendor OUI: 00-25-63
        Vendor PN: LUXN2604BO
        Vendor Rev: B
        Vendor SN: 191400502735
```

liamkearney-msft, 2025-04-04:
@arlakshm for vis

arlakshm, 2025-04-04:
adding more logs
arlakshm@STG01-0101-0400-01T2-lc03:/usr/share/sonic/device/x86_64-arista_7800r3a_36dm2_lc$ show interfaces transceiver info Ethernet96 -n asic0  
Ethernet96: SFP EEPROM detected
        Application Advertisement: N/A
        Connector: Optical Pigtail
        Encoding: 64B/66B
        Extended Identifier: Power Class 4 Module (3.5W max.), No CLEI code present in Page 02h, CDR present in TX, CDR present in RX
        Extended RateSelect Compliance: Rate Select Version 2
        Identifier: Unknown or unspecified
        Length(km): 2.0
        Nominal Bit Rate(100Mbs): 255
        Specification compliance:
                10/40G Ethernet Compliance Code: Extended
                Extended Specification Compliance: 100G PSM4 Parallel SMF
                Fibre Channel Link Length: Unknown
                Fibre Channel Speed: Unknown
                Fibre Channel Transmission Media: Unknown
                Fibre Channel Transmitter Technology: Unknown
                Gigabit Ethernet Compliant Codes: Unknown
                SAS/SATA Compliance Codes: Unknown
                SONET Compliance Codes: Unknown
        Vendor Date Code(YYYY-MM-DD Lot): 2019-04-03   
        Vendor Name: LUXTERA         
        Vendor OUI: 00-25-63
        Vendor PN: LUXN2604BO      
        Vendor Rev: B 
        Vendor SN: 191400501063

arlakshm, 2025-04-04:
```
>>> from sonic_platform.platform import Platform
p = Platform()
platform_chassis = p.get_chassis()

port_num=13
print("{}".format(platform_chassis.get_sfp(port_num).get_xcvr_api()))>>> p = Platform()
>>> platform_chassis = p.get_chassis()
>>> 
>>> port_num=13
>>> print("{}".format(platform_chassis.get_sfp(port_num).get_xcvr_api()))
<sonic_platform_base.sonic_xcvr.api.public.sff8636.Sff8636Api object at 0x7f7936102bd0>
>>> 
```
```
arlakshm@STG01-0101-0400-01T2-lc03:~$ sudo sfputil read-eeprom -n 0 -s 1 -o 128 -p Ethernet96
        00000080 00 
```

arlakshm, 2025-04-04:
root@STG01-0101-0400-01T2-lc03:/# redis-cli -n 0 keys PORT*Ethernet96*
1) "PORT_TABLE:Ethernet96"
```
root@STG01-0101-0400-01T2-lc03:/# redis-cli -n 0 hgetall "PORT_TABLE:Ethernet96"
 1) "mtu"
 2) "9100"
 3) "admin_status"
 4) "up"
 5) "alias"
 6) "Ethernet3/13/1"
 7) "asic_port_name"
 8) "Eth96"
 9) "core_id"
10) "0"
11) "core_port_id"
12) "13"
13) "description"
14) "STG31-0100-0200-02RHE:HundredGigE0/1/0/4"
15) "fec"
16) "rs"
17) "index"
18) "13"
19) "lanes"
20) "40,41,42,43"
21) "macsec"
22) "macsec-profile-two"
23) "num_voq"
24) "8"
25) "pfc_asym"
26) "off"
27) "role"
28) "Ext"
29) "speed"
30) "100000"
31) "tpid"
32) "0x8100"
33) "oper_status"
34) "down"
35) "flap_count"
36) "0"
37) "main"
38) "0x4e,0x4e,0x53,0x4b"
39) "post1"
40) "0xffffffea,0xffffffea,0xffffffea,0xffffffec"
41) "post2"
42) "0x0,0x0,0x0,0x0"
43) "post3"
44) "0x0,0x0,0x0,0x0"
45) "pre1"
46) "0xfffffffb,0xfffffffb,0xfffffffb,0xfffffffb"
47) "pre2"
48) "0x0,0x0,0x0,0x0"
```

arlakshm, 2025-04-04:
@prgeor for viz as well

arlakshm, 2025-04-04:
Logs from EOS

[arista_optics_logs.txt](https://github.com/user-attachments/files/19609746/arista_optics_logs.txt)

kenneth-arista, 2025-04-07:
From our live debug session, the primary root cause is a regression caused by https://github.com/sonic-net/sonic-platform-common/pull/548

prgeor, 2025-04-09:
@arlakshm @kenneth-arista please review the fix here:-

https://github.com/sonic-net/sonic-platform-common/pull/558

Please note this even though is a regression, I have explained in email this is a buggy module which got exposed due to PR548

closed 2025-04-11 by kenneth-arista