aristanetworks/sonic #131: [staging pilot] 100G cable plugged in 400G port caused OA crash

wenyiz2021, 2025-07-07:
we have a staging device running 20240532.28 image, it had an OA crash and seems due to 2025 Jun 30 02:40:14.020251 STG01-0101-0400-01T2-lc03 ERR syncd0#syncd: [06:00.0] SAI_API_PORT:brcm_sai_create_port_serdes:11031 Port lane count 4 is different from supported lane count 8 do you happen to know how to interpret? for more info, some links on this lc were down. 
Plain Text
-------------  -------------------------------  -------  -----  -----  --------------  -------------  ------  -------  -----------------------------------------------  ----------
    Ethernet0          72,73,74,75,76,77,78,79     400G   9100     rs   Ethernet3/1/1   PortChannel1    down       up  QSFP-DD Double Density 8X Pluggable Transceiver         off
    Ethernet8          80,81,82,83,84,85,86,87     400G   9100     rs   Ethernet3/2/1   PortChannel1    down       up  QSFP-DD Double Density 8X Pluggable Transceiver         off
   Ethernet16          88,89,90,91,92,93,94,95     400G   9100     rs   Ethernet3/3/1   PortChannel1      up       up  QSFP-DD Double Density 8X Pluggable Transceiver         off
   Ethernet24      96,97,98,99,100,101,102,103     400G   9100     rs   Ethernet3/4/1   PortChannel1      up       up  QSFP-DD Double Density 8X Pluggable Transceiver         off
   Ethernet32                  104,105,106,107     100G   9100     rs   Ethernet3/5/1   PortChannel5      up       up                                  QSFP28 or later         off
   Ethernet40                  112,113,114,115     100G   9100     rs   Ethernet3/6/1   PortChannel5    down       up                                  QSFP28 or later         off
   Ethernet48                  120,121,122,123     100G   9100     rs   Ethernet3/7/1   PortChannel5      up       up                                  QSFP28 or later         off
   Ethernet56                  128,129,130,131     100G   9100     rs   Ethernet3/8/1   PortChannel5    down       up                                  QSFP28 or later         off
   Ethernet64  136,137,138,139,140,141,142,143     400G   9100     rs   Ethernet3/9/1   PortChannel9      up       up  QSFP-DD Double Density 8X Pluggable Transceiver         off
   Ethernet72          64,65,66,67,68,69,70,71     400G   9100     rs  Ethernet3/10/1   PortChannel9      up       up  QSFP-DD Double Density 8X Pluggable Transceiver         off
   Ethernet80          56,57,58,59,60,61,62,63     400G   9100     rs  Ethernet3/11/1   PortChannel9      up       up  QSFP-DD Double Density 8X Pluggable Transceiver         off
   Ethernet88          48,49,50,51,52,53,54,55     400G   9100     rs  Ethernet3/12/1   PortChannel9    down       up  QSFP-DD Double Density 8X Pluggable Transceiver         off
   Ethernet96                      40,41,42,43     100G   9100     rs  Ethernet3/13/1  PortChannel13      up       up                                  QSFP28 or later         off
  Ethernet104                      32,33,34,35     100G   9100     rs  Ethernet3/14/1  PortChannel13      up       up                                  QSFP28 or later         off
  Ethernet112                      24,25,26,27     100G   9100     rs  Ethernet3/15/1  PortChannel13      up       up                                  QSFP28 or later         off

wenyiz2021, 2025-07-08:
@arista-nwolfe could you assign someone to look at this issue soon and provide an ETA? since this is some image change

rlhui, 2025-07-11:
@arista-nwolfe  any update on this? thanks

wenyiz2021, 2025-08-06:
closing as PR is merged in master and 202405-msft