aristanetworks/sonic #133: [7060] Optics Temperature fluctuating on arista 7060, where 50+ optics are plugged in

judyjoseph, 2025-07-11:
We are trying to understand more on the thermal algorithm and the input parameters for an Arista 7060 pizzabox 

```
admin@str-7060-cx32-1:~$ show version 

SONiC Software Version: SONiC.20241110.22
SONiC OS Version: 12
Distribution: Debian 12.9
Kernel: 6.1.0-29-2-amd64
Build commit: e2b55e6ae8
Build date: Wed Jun 18 08:00:43 UTC 2025
Built by: azureuser@3dccdc78c000000

Platform: x86_64-arista_7060_cx32s
HwSKU: Arista-7060CX-32S-D48C8
ASIC: broadcom
ASIC Count: 1
Serial Number: HBG231701WA
Model Number: DCS-7060CX-32S
```

In the platform.json file we have the thermals defined as below 

```
        "thermals": [
            {
                "name": "Cpu temp sensor",
                "controllable": false
            },
            {
                "name": "Cpu board temp sensor",
                "controllable": false
            },
            {   
                "name": "Back-panel temp sensor",
                "controllable": false
            },  
            {
                "name": "Board sensor",
                "controllable": false
            },
            {
                "name": "Switch chip left sensor",
                "controllable": false
            },
            {   
                "name": "Switch chip right sensor",
                "controllable": false
            },  
            {
                "name": "Front-panel temp sensor",
                "controllable": false
            }
        ],  
```
I did a bit of debugging in the thermalctld daemon in pmon and find that the run_polcy call uses the above thermal_list to call the thermal algorithm which inturn decides on cooling.

```
> /usr/local/bin/thermalctld(887)run()
-> self.thermal_manager.run_policy(self.chassis)
(Pdb) s
--Call--
> /usr/local/lib/python3.11/dist-packages/sonic_platform_base/sonic_thermal_control/thermal_manager_base.py(177)run_policy()
-> @classmethod
(Pdb) list 
172                 policy.validate_duplicate_policy(cls._policy_dict.values())
173                 cls._policy_dict[name] = policy
174             else:
175                 raise Exception('{} not found in policy'.format(cls.JSON_FIELD_POLICY_NAME))
176  
(Pdb) chassis
<arista.utils.sonic_platform.chassis.Chassis object at 0x7f8d232fbf10>
(Pdb) dir(chassis)
['DEVICE_TYPE', 'POLL_INTERVAL', 'REBOOT_CAUSE_DICT', 'REBOOT_CAUSE_HARDWARE_BIOS', 'REBOOT_CAUSE_HARDWARE_BUTTON', 'REBOOT_CAUSE_HARDWARE_CPU', 'REBOOT_CAUSE_HARDWARE_OTHER', 'REBOOT_CAUSE_HARDWARE_RESET_FROM_ASIC', 'REBOOT_CAUSE_INSUFFICIENT_FAN_SPEED', 'REBOOT_CAUSE_NON_HARDWARE', 'REBOOT_CAUSE_POWER_LOSS', 'REBOOT_CAUSE_THERMAL_OVERLOAD_ASIC', 'REBOOT_CAUSE_THERMAL_OVERLOAD_CPU', 'REBOOT_CAUSE_THERMAL_OVERLOAD_OTHER', 'REBOOT_CAUSE_WATCHDOG', 'STATUS_LED_COLOR_AMBER', 'STATUS_LED_COLOR_GREEN', 'STATUS_LED_COLOR_OFF', 'STATUS_LED_COLOR_RED', '__class__', '__delattr__', '__dict__', '__dir__', '__doc__', '__eq__', '__format__', '__ge__', '__getattribute__', '__getstate__', '__gt__', '__hash__', '__init__', '__init_subclass__', '__le__', '__lt__', '__module__', '__ne__', '__new__', '__reduce__', '__reduce_ex__', '__repr__', '__setattr__', '__sizeof__', '__str__', '__subclasshook__', '__weakref__', '_chassis', '_component_list', '_current_sensor_list', '_eeprom', '_event_watcher', '_fan_drawer_list', '_fan_list', '_get_event_watcher', '_inventory', '_module_list', '_platform', '_psu_list', '_sfp_list', '_status_led', '_thermal_list', '_voltage_sensor_list', '_watchdog', 'getPlatform', 'getThermalControl', 'get_all_components', 'get_all_current_sensors', 'get_all_fan_drawers', 'get_all_fans', 'get_all_modules', 'get_all_psus', 'get_all_sfps', 'get_all_thermals', 'get_all_voltage_sensors', 'get_base_mac', 'get_change_event', 'get_component', 'get_controlplane_state', 'get_current_sensor', 'get_dataplane_state', 'get_dpu_id', 'get_eeprom', 'get_fan', 'get_fan_drawer', 'get_model', 'get_module', 'get_module_index', 'get_my_slot', 'get_name', 'get_num_components', 'get_num_current_sensors', 'get_num_fan_drawers', 'get_num_fans', 'get_num_modules', 'get_num_psus', 'get_num_sfps', 'get_num_thermals', 'get_num_voltage_sensors', 'get_port_or_cage_type', 'get_position_in_parent', 'get_presence', 'get_psu', 'get_reboot_cause', 'get_revision', 'get_serial', 'get_serial_number', 'get_sfp', 'get_status', 'get_status_led', 'get_supervisor_slot', 'get_system_eeprom_info', 'get_thermal', 'get_thermal_manager', 'get_uid_led', 'get_voltage_sensor', 'get_watchdog', 'init_midplane_switch', 'initizalize_system_led', 'is_dpu', 'is_modular_chassis', 'is_replaceable', 'is_smartswitch', 'set_status_led', 'set_uid_led']
(Pdb) chassis._thermal_list
[<arista.utils.sonic_platform.thermal.Thermal object at 0x7f8d219b3990>, <arista.utils.sonic_platform.thermal.Thermal object at 0x7f8d219b39d0>, <arista.utils.sonic_platform.thermal.Thermal object at 0x7f8d219b3a10>, <arista.utils.sonic_platform.thermal.Thermal object at 0x7f8d219b3a50>, <arista.utils.sonic_platform.thermal.Thermal object at 0x7f8d219b3a90>, <arista.utils.sonic_platform.thermal.Thermal object at 0x7f8d219b3b10>, <arista.utils.sonic_platform.thermal.Thermal object at 0x7f8d219b3b50>]
```

We have a few questions to see how the optics temperature can be a factor in cooling 

1. how the optics temperature is included in this thermal_list
2. Does any of the above sensors defined in "thermals" correlate to the optics temperature?
3. The optics temperature is recorded in the DOM_SESOR table, are we using is directly or indirectly for cooling ?

Please share more details on the above

Staphylo, 2025-07-15:
I believe there's also an email thread ongoing about this topic.
To answer your specific questions here:

1) The optics are not part of `ChassisBase._thermal_list` only `Thermal` objects specific to the product are.
The xcvrs are available under `ChassisBase._sfp_list` and they have a specific way to get the temperature out of them. Because this is going beyond the hardware platform API and falling into the realm of xcvr management we do no query xcvrs directly.  Instead we rely on `xcvrd` to populate the necessary data into the DB.
If we were to query temperature directly, it would look like `ChassisBase.get_sfp(1).get_xcvr_api().get_temperature()`
Note that we have a knob to do that https://github.com/aristanetworks/sonic/blob/b275267ad2b70a8d57abbb835c7c286d918a3295/arista/utils/sonic_platform/thermal_helper.py#L397

2) We are using the data from both `TRANSCEIVER_DOM_SENSOR` and `TRANSCEIVER_DOM_THRESHOLD` tables.
I would suggest you to read through the `thermal_helper.py` file linked above.
It is responsible for gathering data from multiple places and feed the algorithm with it.

3) I would say that the algorithm is closer to independent mode rather than dependent.
It considers all sensors it can access and then decides the fan speed.
It is not considering any other factor such as system load, ambiant temperature, optic power draw, ...

judyjoseph, 2025-07-18:
@Staphylo I find a few values which is used in the cooling algorithm 

```
         cls.instance_.cooling_data_points = 10
         cls.instance_.cooling_export_path = None
         cls.instance_.cooling_max_decrease = 10
         cls.instance_.cooling_max_increase = 25
         cls.instance_.cooling_min_speed = 30
         cls.instance_.cooling_loop_interval = 20
         cls.instance_.cooling_target_offset = 0
         cls.instance_.cooling_target_factor = 0.8
         cls.instance_.cooling_gc_count = 15
```

https://github.com/aristanetworks/sonic/blob/b275267ad2b70a8d57abbb835c7c286d918a3295/arista/core/config.py#L50

Atleast one of which I found was this variable **cooling_gc_count** which was deciding how frequently the xcvr and other thermal info data should be updated to be used in the cooling algorithm. So even if the xcvrd was able to get DOM sensor temp data faster, the data is not really getting used in cooling algorithm until we reduce this from 15 to like 1 (to make it update every 20sec).  Please correct me if I am wrong.

Also share your thoughts on this and on other parameters tuning any of which could improve the cooling algo.

Staphylo, 2025-07-21:
@judyjoseph you're correct, there is an issue here which is actually a bug.
The call to `self.update()` under `gc()` happens after the condition on whether to run the garbage collection for outdated objects.
Instead it was intended for the `self.update()` call to happen at every iteration of the algorithm, so it should be moved before the condition.
We'll work on a fix and get back to you.

In the meantime it's perfectly fine to set this value to  `1` as a mitigation.
Note that all these settings can be saved under `/etc/sonic/arista.config` such as
```
cooling_gc_count: 1
```
In this case, you'll just need to restart thermalctld and it should pick up the new value `docker exec pmon supervisorctl restart thermactld`

judyjoseph, 2025-07-22:
Hi @Staphylo another observation - It takes more time to update the fan speed

```
DEBUG: ThermalInfo(thermal=CoolingThermal(Cpu temp sensor), value=35.0, target=70.0, overheat=95.0, delta=-35.0, deltap=-1.4)
DEBUG: ThermalInfo(thermal=CoolingThermal(Ambient), value=32.375, target=55.0, overheat=75.0, delta=-22.625, deltap=-1.13125)
DEBUG: ThermalInfo(thermal=CoolingThermal(Outlet), value=23.187, target=65.0, overheat=80.0, delta=-41.813, deltap=-2.7875333333333336)
DEBUG: ThermalInfo(thermal=CoolingThermal(Switch Card temp sensor), value=30.5, target=85.0, overheat=95.0, delta=-54.5, deltap=-5.45)
DEBUG: ThermalInfo(thermal=CoolingThermal(Air Exit Behind TH5), value=37.25, target=85.0, overheat=95.0, delta=-47.75, deltap=-4.775)
DEBUG: ThermalInfo(thermal=CoolingThermal(Left Edge PCB Near Rear of Switch), value=23.625, target=85.0, overheat=95.0, delta=-61.375, deltap=-6.1375)
DEBUG: ThermalInfo(thermal=CoolingThermal(Air Inlet), value=24.875, target=85.0, overheat=95.0, delta=-60.125, deltap=-6.0125)
DEBUG: ThermalInfo(thermal=CoolingThermal(Management Card Inlet), value=21.062, target=85.0, overheat=95.0, delta=-63.938, deltap=-6.393800000000001)
DEBUG: ThermalInfo(thermal=CoolingThermal(Power supply 1 inlet sensor), value=23.625, target=60.0, overheat=65.0, delta=-36.375, deltap=-7.275)
DEBUG: ThermalInfo(thermal=CoolingThermal(Power supply 1 secondary hotspot sensor), value=33.0, target=110.0, overheat=120.0, delta=-77.0, deltap=-7.7)
DEBUG: ThermalInfo(thermal=CoolingThermal(Power supply 1 primary hotspot sensor), value=43.375, target=110.0, overheat=115.0, delta=-66.625, deltap=-13.325)
DEBUG: ThermalInfo(thermal=CoolingThermal(Power supply 2 inlet sensor), value=24.0, target=60.0, overheat=65.0, delta=-36.0, deltap=-7.2)
DEBUG: ThermalInfo(thermal=CoolingThermal(Power supply 2 secondary hotspot sensor), value=28.125, target=110.0, overheat=120.0, delta=-81.875, deltap=-8.1875)
DEBUG: ThermalInfo(thermal=CoolingThermal(Power supply 2 primary hotspot sensor), value=40.375, target=110.0, overheat=115.0, delta=-69.625, deltap=-13.925)
DEBUG: ThermalInfo(thermal=CoolingXcvrThermal(Ethernet128), value=47.688, target=60.0, overheat=75.0, delta=-12.311999999999998, deltap=-0.8207999999999999)
DEBUG: ThermalInfo(thermal=CoolingXcvrThermal(Ethernet288), value=38.48, target=60.0, overheat=75.0, delta=-21.520000000000003, deltap=-1.434666666666667)
DEBUG: ThermalInfo(thermal=CoolingXcvrThermal(Ethernet216), value=46.52, target=60.0, overheat=75.0, delta=-13.479999999999997, deltap=-0.8986666666666665)
DEBUG: ThermalInfo(thermal=CoolingXcvrThermal(Ethernet164), value=49.219, target=60.0, overheat=75.0, delta=-10.780999999999999, deltap=-0.7187333333333332)
DEBUG: ThermalInfo(thermal=CoolingXcvrThermal(Ethernet80), value=51.41, target=60.0, overheat=75.0, delta=-8.590000000000003, deltap=-0.5726666666666669)
DEBUG: ThermalInfo(thermal=CoolingXcvrThermal(Ethernet316), value=39.402, target=60.0, overheat=75.0, delta=-20.598, deltap=-1.3732)
DEBUG: ThermalInfo(thermal=CoolingXcvrThermal(Ethernet224), value=47.898, target=60.0, overheat=75.0, delta=-12.101999999999997, deltap=-0.8067999999999997)
DEBUG: ThermalInfo(thermal=CoolingXcvrThermal(Ethernet136), value=55.691, target=60.0, overheat=75.0, delta=-4.3089999999999975, deltap=-0.2872666666666665)
DEBUG: ThermalInfo(thermal=CoolingXcvrThermal(Ethernet92), value=45.926, target=60.0, overheat=75.0, delta=-14.073999999999998, deltap=-0.9382666666666666)
DEBUG: ThermalInfo(thermal=CoolingXcvrThermal(Ethernet264), value=52.84, target=60.0, overheat=75.0, delta=-7.159999999999997, deltap=-0.4773333333333331)
DEBUG: ThermalInfo(thermal=CoolingXcvrThermal(Ethernet212), value=50.32, target=60.0, overheat=75.0, delta=-9.68, deltap=-0.6453333333333333)
DEBUG: ThermalInfo(thermal=CoolingXcvrThermal(Ethernet252), value=45.281, target=60.0, overheat=75.0, delta=-14.719000000000001, deltap=-0.9812666666666667)
DEBUG: ThermalInfo(thermal=CoolingXcvrThermal(Ethernet284), value=32.824, target=60.0, overheat=75.0, delta=-27.176000000000002, deltap=-1.8117333333333334)
DEBUG: ThermalInfo(thermal=CoolingXcvrThermal(Ethernet248), value=44.938, target=60.0, overheat=75.0, delta=-15.061999999999998, deltap=-1.004133333333333)
DEBUG: ThermalInfo(thermal=CoolingXcvrThermal(Ethernet188), value=46.465, target=60.0, overheat=75.0, delta=-13.534999999999997, deltap=-0.9023333333333331)
DEBUG: ThermalInfo(thermal=CoolingXcvrThermal(Ethernet296), value=40.445, target=60.0, overheat=75.0, delta=-19.555, deltap=-1.3036666666666668)
DEBUG: ThermalInfo(thermal=CoolingXcvrThermal(Ethernet220), value=46.348, target=60.0, overheat=75.0, delta=-13.652000000000001, deltap=-0.9101333333333333)
DEBUG: ThermalInfo(thermal=CoolingXcvrThermal(Ethernet280), value=30.957, target=60.0, overheat=75.0, delta=-29.043, deltap=-1.9362)
DEBUG: ThermalInfo(thermal=CoolingXcvrThermal(Ethernet256), value=48.109, target=60.0, overheat=75.0, delta=-11.890999999999998, deltap=-0.7927333333333332)
DEBUG: ThermalInfo(thermal=CoolingXcvrThermal(Ethernet60), value=50.855, target=60.0, overheat=75.0, delta=-9.145000000000003, deltap=-0.6096666666666669)
DEBUG: ThermalInfo(thermal=CoolingXcvrThermal(Ethernet84), value=52.289, target=60.0, overheat=75.0, delta=-7.7109999999999985, deltap=-0.5140666666666666)
DEBUG: ThermalInfo(thermal=CoolingXcvrThermal(Ethernet148), value=51.055, target=60.0, overheat=75.0, delta=-8.945, deltap=-0.5963333333333334)
[... cut ...]
   slot1  green    fan1      41%      exhaust     Present        OK  20250722 17:23:19
   slot2  green    fan2      41%      exhaust     Present        OK  20250722 17:23:19
   slot3  green    fan3      41%      exhaust     Present        OK  20250722 17:23:19
   slot4  green    fan4      41%      exhaust     Present        OK  20250722 17:23:19
```

Another point from code reading : 
  This flag self.overheat (https://github.com/aristanetworks/sonic/blob/e052a11d944e68ef8ae59f2040068fa84809447b/arista/core/cooling.py#L154) never gets used ? Can you check

judyjoseph, 2025-07-24:
Thanks [@Samuel Angebault](mailto:staphylo@arista.com) for the meeting, noting down a few points discussed 

**Cooling Algo:**
1.	Two fixes which helped
a.	Set cooling_gc_count = 1. This is needed for the cooling algo use the optics temp values in every polling cycle of 20sec (https://github.com/aristanetworks/sonic/blob/b275267ad2b70a8d57abbb835c7c286d918a3295/arista/core/config.py#L58
b.	Don’t skip the fan speed change if the delta/deltap > 0, when current optics temp is > target value (https://github.com/aristanetworks/sonic/blob/e052a11d944e68ef8ae59f2040068fa84809447b/arista/core/cooling.py#L219)
2.	The PID (Proportional–Integral–Derivative) approach used in EOS might be difficult to bring it in SONiC.
Samuel is doing more testing with the above fixes identified and any more optimization in cooling algorithm – to share it soon

**Xcvr dom mgr polling:**
1.	As identified earlier we can reduce the interval between polling from 60sec  5sec (https://github.com/sonic-net/sonic-platform-daemons/blob/c8856950f38f9d9275866e24e449c81e0a7b09f1/sonic-xcvrd/xcvrd/dom/dom_mgr.py#L36
2.	In the 7060 in lab with 70+ optics - I find that the VDM collection logic takes ~60sec for all the interfaces. I have have disabled it - which resulted in dom sensor/temp data to be populated in DB every 60sec – helps cooling algorithm to kick in early.

       We can do some optimizations here to get DOM data ( may be these are already known/planned - Prince/Mihir )
          a)	Can we have the dom_mgr polling as a separate process ? e.g., use multi-processing python lib instead of threading lib.
          b)	Can we optimize DOM/VDM fetch by doing parallel i2c access – Samuel mentioned in Arista we have one i2c controller per 8 optics, but this could be platform dependent.

judyjoseph, 2025-08-25:
@Staphylo could you confirm if the changes in the cooling algorithm are completed. I will create a separate issue for the xcvrd DOM sensor data poll.

Staphylo, 2025-08-28:
@judyjoseph the commit I linked above is moving the `self.update()` call before the check for the gc as was initially intended.
There is therefore no need to change `cooling_gc_count`

Staphylo, 2025-08-29:
@judyjoseph I just thought about it now, but out of curiosity did you try setting `cooling_xcvrs_via_api: true`.
Setting this property means that the cooling algorithm will poll the temperature information directly from the xcvrs rather than pulling it from the database.
This should ensure that every iteration is running on latest temperature information.