aristanetworks/sonic #134: Arista SONiC T2[cph07-0101-0100-03t2-lc04]: running sonic.20240532.28, LC unexpected reboot

deepak-singhal0408, 2025-07-18:
This is regarding issue seen in Production Pilot of Arista SONiC T2 chassis.. 
Today we hit an unexpected reboot of one of the linecard in one of the converted chassis...

Image version : 20240532.28.. 
TAC CASE#  701822
 
 2025 Jul 18 16:02:16.890624 CPH07-0101-0100-03T2-lc04 INFO kernel: [    4.447764] BERT: Error records from previous boot:
2025 Jul 18 16:02:16.890625 CPH07-0101-0100-03T2-lc04 INFO kernel: [    4.447766] [Hardware Error]: event severity: info
2025 Jul 18 16:02:16.890625 CPH07-0101-0100-03T2-lc04 INFO kernel: [    4.447767] [Hardware Error]:  Error 0, type: fatal
2025 Jul 18 16:02:16.890626 CPH07-0101-0100-03T2-lc04 INFO kernel: [    4.447768] [Hardware Error]:  fru_text: ProcessorError
2025 Jul 18 16:02:16.890626 CPH07-0101-0100-03T2-lc04 INFO kernel: [    4.447769] [Hardware Error]:   section_type: IA32/X64 processor error
2025 Jul 18 16:02:16.890627 CPH07-0101-0100-03T2-lc04 INFO kernel: [    4.447770] [Hardware Error]:   Local APIC_ID: 0x0
2025 Jul 18 16:02:16.890632 CPH07-0101-0100-03T2-lc04 INFO kernel: [    4.447771] [Hardware Error]:   CPUID Info:
2025 Jul 18 16:02:16.890632 CPH07-0101-0100-03T2-lc04 INFO kernel: [    4.447773] [Hardware Error]:   00000000: 00800f12 00000000 00080800 00000000
2025 Jul 18 16:02:16.890633 CPH07-0101-0100-03T2-lc04 INFO kernel: [    4.447775] [Hardware Error]:   00000010: 76d8320b 00000000 178bfbff 00000000
2025 Jul 18 16:02:16.890633 CPH07-0101-0100-03T2-lc04 INFO kernel: [    4.447776] [Hardware Error]:   00000020: 48ab7f57 4f6cdc34 b5b0d3a7 1443a7b0
2025 Jul 18 16:02:16.890634 CPH07-0101-0100-03T2-lc04 INFO kernel: [    4.447777] [Hardware Error]:   Error Information Structure 0:
2025 Jul 18 16:02:16.890635 CPH07-0101-0100-03T2-lc04 INFO kernel: [    4.447778] [Hardware Error]:    Error Structure Type: unknown
2025 Jul 18 16:02:16.890635 CPH07-0101-0100-03T2-lc04 INFO kernel: [    4.447779] [Hardware Error]:    Error Structure Type: 00000001-0000-0000-0700-180000000000
2025 Jul 18 16:02:16.890641 CPH07-0101-0100-03T2-lc04 INFO kernel: [    4.447781] [Hardware Error]:  Error 1, type: fatal
2025 Jul 18 16:02:16.890642 CPH07-0101-0100-03T2-lc04 INFO kernel: [    4.447782] [Hardware Error]:  fru_text: ProcessorError
2025 Jul 18 16:02:16.890643 CPH07-0101-0100-03T2-lc04 INFO kernel: [    4.447782] [Hardware Error]:   section_type: IA32/X64 processor error
2025 Jul 18 16:02:16.890643 CPH07-0101-0100-03T2-lc04 INFO kernel: [    4.447783] [Hardware Error]:   Local APIC_ID: 0x2
2025 Jul 18 16:02:16.890644 CPH07-0101-0100-03T2-lc04 INFO kernel: [    4.447784] [Hardware Error]:   CPUID Info:
2025 Jul 18 16:02:16.890644 CPH07-0101-0100-03T2-lc04 INFO kernel: [    4.447785] [Hardware Error]:   00000000: 00800f12 00000000 02080800 00000000
2025 Jul 18 16:02:16.890649 CPH07-0101-0100-03T2-lc04 INFO kernel: [    4.447786] [Hardware Error]:   00000010: 76d8320b 00000000 178bfbff 00000000
2025 Jul 18 16:02:16.890649 CPH07-0101-0100-03T2-lc04 INFO kernel: [    4.447787] [Hardware Error]:   00000020: a55701f5 43dee3ef 9b2472ac 2cad3f57
2025 Jul 18 16:02:16.890650 CPH07-0101-0100-03T2-lc04 INFO kernel: [    4.447788] [Hardware Error]:   Error Information Structure 0:
2025 Jul 18 16:02:16.890651 CPH07-0101-0100-03T2-lc04 INFO kernel: [    4.447789] [Hardware Error]:    Error Structure Type: unknown
2025 Jul 18 16:02:16.890651 CPH07-0101-0100-03T2-lc04 INFO kernel: [    4.447790] [Hardware Error]:    Error Structure Type: 00000001-0000-0000-1f00-020600000000
2025 Jul 18 16:02:16.890652 CPH07-0101-0100-03T2-lc04 INFO kernel: [    4.447791] [Hardware Error]:  Error 2, type: fatal
2025 Jul 18 16:02:16.890655 CPH07-0101-0100-03T2-lc04 INFO kernel: [    4.447791] [Hardware Error]:  fru_text: ProcessorError
2025 Jul 18 16:02:16.890657 CPH07-0101-0100-03T2-lc04 INFO kernel: [    4.447792] [Hardware Error]:   section_type: IA32/X64 processor error
2025 Jul 18 16:02:16.890658 CPH07-0101-0100-03T2-lc04 INFO kernel: [    4.447793] [Hardware Error]:   Local APIC_ID: 0xb
2025 Jul 18 16:02:16.890658 CPH07-0101-0100-03T2-lc04 INFO kernel: [    4.447793] [Hardware Error]:   CPUID Info:
2025 Jul 18 16:02:16.890659 CPH07-0101-0100-03T2-lc04 INFO kernel: [    4.447794] [Hardware Error]:   00000000: 00800f12 00000000 0b080800 00000000
2025 Jul 18 16:02:16.890659 CPH07-0101-0100-03T2-lc04 INFO kernel: [    4.447796] [Hardware Error]:   00000010: 76d8320b 00000000 178bfbff 00000000
2025 Jul 18 16:02:16.890660 CPH07-0101-0100-03T2-lc04 INFO kernel: [    4.447797] [Hardware Error]:   00000020: a55701f5 43dee3ef 9b2472ac 2cad3f57
2025 Jul 18 16:02:16.890664 CPH07-0101-0100-03T2-lc04 INFO kernel: [    4.447798] [Hardware Error]:   Error Information Structure 0:
2025 Jul 18 16:02:16.890665 CPH07-0101-0100-03T2-lc04 INFO kernel: [    4.447798] [Hardware Error]:    Error Structure Type: unknown
2025 Jul 18 16:02:16.890666 CPH07-0101-0100-03T2-lc04 INFO kernel: [    4.447799] [Hardware Error]:    Error Structure Type: 00000001-0000-0000-1f00-060600000000
2025 Jul 18 16:02:16.890666 CPH07-0101-0100-03T2-lc04 INFO kernel: [    4.447800] BERT: Total records found: 1

mDEEPSINGHAL@CPH07-0101-0100-03T2-lc04:/var/log$ show reboot-cause hist
Name                 Cause                                                                                                   Time    User    Comment
-------------------  ------------------------------------------------------------------------------------------------------  ------  ------  -------------------------------------------------
2025_07_18_16_02_27  Thermal Overload: Other (overtemp, description: detailed fault powerup=115, time: 2025-07-18 19:26:19)  N/A     N/A     Unknown
2025_07_04_01_35_29  Watchdog (watchdog, description: Watchdog fired, time: 2025-07-04 01:21:22)                             N/A     N/A     Unknown (First boot of SONiC version 20240532.28)

mDEEPSINGHAL@CPH07-0101-0100-03T2-lc04:~$ show plat sum
Platform: x86_64-arista_7800r3a_36dm2_lc
HwSKU: Arista-7800R3A-36DM2-D36
ASIC: broadcom
ASIC Count: 2
Serial Number: SGD2417037P
Model Number: 7800R3A-36DM2-LC
Hardware Revision: 2b.0a
Switch Type: voq
DEEPSINGHAL@CPH07-0101-0100-03T2-lc04:~$ fwutil show status
Error: Root privileges are required. Aborting...
Aborted!
DEEPSINGHAL@CPH07-0101-0100-03T2-lc04:~$ sudo fwutil show status
Chassis           Module    Component               Version                         Description
----------------  --------  ----------------------  ------------------------------  ---------------------
7800R3A-36DM2-LC            Aboot()                 Aboot-norcal10-10.0.2-26144132  Bootloader
                            Scd(addr=0000:00:18.7)  0x0                             System Control Device
                            Scd(addr=0000:08:00.0)  0x30100                         System Control Device
DEEPSINGHAL@CPH07-0101-0100-03T2-lc04:~$ 
DEEPSINGHAL@CPH07-0101-0100-03T2-lc04:~$ 
DEEPSINGHAL@CPH07-0101-0100-03T2-lc04:~$ 
DEEPSINGHAL@CPH07-0101-0100-03T2-lc04:~$ cd /var/log
DEEPSINGHAL@CPH07-0101-0100-03T2-lc04:/var/log$ ls
arista-daemon.log  audit     btmp    cron.log  gnmi.log  kern.log.1  lost+found  private  swss    syslog.1     syslog.3.gz  syslog.5.gz  teamd.log      user.log    wtmp
arista.log         auth.log  btmp.1  frr       kern.log  lastlog     ntpsec      README   syslog  syslog.2.gz  syslog.4.gz  sysstat      telemetry.log  user.log.1  wtmp.1
DEEPSINGHAL@CPH07-0101-0100-03T2-lc04:/var/log$ show kdump config
Kdump administrative mode: Disabled
Kdump operational mode: Not Ready
Kdump memory reservation: 0M-2G:256M,2G-4G:320M,4G-8G:384M,8G-:448M
Maximum number of Kdump files: 3

deepak-singhal0408, 2025-07-18:
Techsupport: [https://microsoft.sharepoint.com/:f:/t/AristaSONiCWorkingGroup/Eh3wiuoy6WJGplDge_SW4cABLsOmlm8Ab-YBnAjX19e9IQ?e=Xz8IJX](https://microsoft.sharepoint.com/:f:/t/AristaSONiCWorkingGroup/Eh3wiuoy6WJGplDge_SW4cABLsOmlm8Ab-YBnAjX19e9IQ?e=Xz8IJX&xsdata=MDV8MDJ8fGZkNDAwZDBhM2Q2NDQ1YTFkYTdkMDhkZGM2NDllZDRlfDcyZjk4OGJmODZmMTQxYWY5MWFiMmQ3Y2QwMTFkYjQ3fDB8MHw2Mzg4ODQ3NDI5NzI3NTg2NTN8VW5rbm93bnxWR1ZoYlhOVFpXTjFjbWwwZVZObGNuWnBZMlY4ZXlKRFFTSTZJbFJsWVcxelgwRlVVRk5sY25acFkyVmZVMUJQVEU5R0lpd2lWaUk2SWpBdU1DNHdNREF3SWl3aVVDSTZJbGRwYmpNeUlpd2lRVTRpT2lKUGRHaGxjaUlzSWxkVUlqb3hNWDA9fDF8TDJOb1lYUnpMekU1T20xbFpYUnBibWRmV1hwT2JGcFVhelZOTWtWMFdtcGthVmw1TURCUFJHaHRURmRLYlZscVdYUk5la3B0V1hwWk5FNTZRbTFPUkZwcVFIUm9jbVZoWkM1Mk1pOXRaWE56WVdkbGN5OHhOelV5T0RjM05EazJOakU1fGI3YjdkYzgwMzAzMDQ1MDM3ODJmMDhkZGM2NDllZDRjfGE5YTMyNjUwYjNkOTRkNWI4OTQ2MWIxNGJlMjJiYjI5&sdata=TEwrN2MzVFB3cHY4cURUR21EVDdyQ2FFNG1uK29BOGI1azRjczRKV3FWMD0%3D&ovuser=72f988bf-86f1-41af-91ab-2d7cd011db47%2Cdeepsinghal%40microsoft.com)

deepak-singhal0408, 2025-07-18:
@ritahui

deepak-singhal0408, 2025-07-22:
BERT File Uploaded.. https://microsoft.sharepoint.com/:u:/t/AristaSONiCWorkingGroup/EetmyFhcl8xBjfoe6oxEd30B-GnvxxW5WfjW0lTDmOVSKQ?e=PG39We

kenneth-arista, 2025-08-05:
As discussed offline. Let's continue to monitor. Meanwhile closing this issue.

closed 2025-08-05 by kenneth-arista

kenneth-arista, 2025-08-15:
Some historical details:
* KDUMP was not enabled on the LC and thus the context (e.g. backtrace) of the reboot was missing
* Asked to enable KDUMP and to continue to monitor subsequently
* The kernel BERT info wasn't useful because the LC was rebooted again so that any last boot info was lost
* Syslogs for the LC and supervisor were examined and nothing obvious hinted at why the LC restarted