aristanetworks/sonic #136: [Prod] pmon exited with stacktrace

wenyiz2021, 2025-07-29:
On 202405 image, pmon exited unexpectedly and immediately came back. opening this issue to understand the stacktrace.

2025 Jul 24 13:56:30.457959 CPH07-0101-0100-08T2-sup00 NOTICE syncd3#syncd: :- threadFunction: time span 24 ms for 'get:SAI_OBJECT_TYPE_PORT:oid:0x30100000000002d'
2025 Jul 24 13:56:30.888140 CPH07-0101-0100-08T2-sup00 NOTICE syncd4#syncd: :- threadFunction: time span 4 ms for 'get:SAI_OBJECT_TYPE_PORT:oid:0x40100000000007f'
2025 Jul 24 13:56:56.240627 CPH07-0101-0100-08T2-sup00 INFO pmon#supervisord: pcied Traceback (most recent call last):
2025 Jul 24 13:56:56.240627 CPH07-0101-0100-08T2-sup00 INFO pmon#supervisord: pcied   File "/usr/local/bin/pcied", line 218, in <module>
2025 Jul 24 13:56:56.240754 CPH07-0101-0100-08T2-sup00 INFO pmon#supervisord: pcied     sys.exit(main())
2025 Jul 24 13:56:56.240754 CPH07-0101-0100-08T2-sup00 INFO pmon#supervisord: pcied              ^^^^^^
2025 Jul 24 13:56:56.240779 CPH07-0101-0100-08T2-sup00 INFO pmon#supervisord: pcied   File "/usr/local/bin/pcied", line 210, in main
2025 Jul 24 13:56:56.240788 CPH07-0101-0100-08T2-sup00 INFO pmon#supervisord: pcied     while pcied.run():
2025 Jul 24 13:56:56.240804 CPH07-0101-0100-08T2-sup00 INFO pmon#supervisord: pcied           ^^^^^^^^^^^
2025 Jul 24 13:56:56.240804 CPH07-0101-0100-08T2-sup00 INFO pmon#supervisord: pcied   File "/usr/local/bin/pcied", line 197, in run
2025 Jul 24 13:56:56.240865 CPH07-0101-0100-08T2-sup00 INFO pmon#supervisord: pcied     self.check_pcie_devices()
2025 Jul 24 13:56:56.240865 CPH07-0101-0100-08T2-sup00 INFO pmon#supervisord: pcied   File "/usr/local/bin/pcied", line 156, in check_pcie_devices
2025 Jul 24 13:56:56.240894 CPH07-0101-0100-08T2-sup00 INFO pmon#supervisord: pcied     self.resultInfo = platform_pcieutil.get_pcie_check()
2025 Jul 24 13:56:56.240927 CPH07-0101-0100-08T2-sup00 INFO pmon#supervisord: pcied                       ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
2025 Jul 24 13:56:56.240927 CPH07-0101-0100-08T2-sup00 INFO pmon#supervisord: pcied   File "/usr/lib/python3/dist-packages/arista/utils/sonic_platform/pcie.py", line 94, in get_pcie_check
2025 Jul 24 13:56:56.243003 CPH07-0101-0100-08T2-sup00 INFO pmon#supervisord: pcied     devices = self.get_pcie_device()
2025 Jul 24 13:56:56.243106 CPH07-0101-0100-08T2-sup00 INFO pmon#supervisord: pcied               ^^^^^^^^^^^^^^^^^^^^^^
2025 Jul 24 13:56:56.243106 CPH07-0101-0100-08T2-sup00 INFO pmon#supervisord: pcied   File "/usr/lib/python3/dist-packages/arista/utils/sonic_platform/pcie.py", line 85, in get_pcie_device
2025 Jul 24 13:56:56.243146 CPH07-0101-0100-08T2-sup00 INFO pmon#supervisord: pcied     for port in self.iterPciPorts():
2025 Jul 24 13:56:56.243242 CPH07-0101-0100-08T2-sup00 INFO pmon#supervisord: pcied   File "/usr/lib/python3/dist-packages/arista/utils/sonic_platform/pcie.py", line 32, in iterPciPorts
2025 Jul 24 13:56:56.243242 CPH07-0101-0100-08T2-sup00 INFO pmon#supervisord: pcied     chassis.loadAll()
2025 Jul 24 13:56:56.243282 CPH07-0101-0100-08T2-sup00 INFO pmon#supervisord: pcied   File "/usr/lib/python3/dist-packages/arista/core/modular.py", line 95, in loadAll
2025 Jul 24 13:56:56.246777 CPH07-0101-0100-08T2-sup00 INFO pmon#supervisord: pcied     self.loadLinecards()
2025 Jul 24 13:56:56.246777 CPH07-0101-0100-08T2-sup00 INFO pmon#supervisord: pcied   File "/usr/lib/python3/dist-packages/arista/core/modular.py", line 76, in loadLinecards
2025 Jul 24 13:56:56.246842 CPH07-0101-0100-08T2-sup00 INFO pmon#supervisord: pcied     slot.loadCard(standbyOnly=standbyOnly)
2025 Jul 24 13:56:56.246860 CPH07-0101-0100-08T2-sup00 INFO pmon#supervisord: pcied   File "/usr/lib/python3/dist-packages/arista/core/card.py", line 136, in loadCard
2025 Jul 24 13:56:56.247276 CPH07-0101-0100-08T2-sup00 INFO pmon#supervisord: pcied     eeprom = self.getEeprom()
2025 Jul 24 13:56:56.247405 CPH07-0101-0100-08T2-sup00 INFO pmon#supervisord: pcied              ^^^^^^^^^^^^^^^^
2025 Jul 24 13:56:56.247422 CPH07-0101-0100-08T2-sup00 INFO pmon#supervisord: pcied   File "/usr/lib/python3/dist-packages/arista/components/denali/card.py", line 257, in getEeprom
2025 Jul 24 13:56:56.247771 CPH07-0101-0100-08T2-sup00 INFO pmon#supervisord: pcied     self.card.setupIdentification()
2025 Jul 24 13:56:56.247888 CPH07-0101-0100-08T2-sup00 INFO pmon#supervisord: pcied   File "/usr/lib/python3/dist-packages/arista/components/denali/card.py", line 233, in setupIdentification
2025 Jul 24 13:56:56.247905 CPH07-0101-0100-08T2-sup00 INFO pmon#supervisord: pcied     self.eeprom.setup()
2025 Jul 24 13:56:56.247991 CPH07-0101-0100-08T2-sup00 INFO pmon#supervisord: pcied   File "/usr/lib/python3/dist-packages/arista/core/component/__init__.py", line 103, in setup
2025 Jul 24 13:56:56.248363 CPH07-0101-0100-08T2-sup00 INFO pmon#supervisord: pcied     driver.setup()
2025 Jul 24 13:56:56.248477 CPH07-0101-0100-08T2-sup00 INFO pmon#supervisord: pcied   File "/usr/lib/python3/dist-packages/arista/drivers/eeprom.py", line 23, in setup
2025 Jul 24 13:56:56.248806 CPH07-0101-0100-08T2-sup00 INFO pmon#supervisord: pcied     waitFor(lambda: os.path.exists(self.eepromPath()),
2025 Jul 24 13:56:56.248826 CPH07-0101-0100-08T2-sup00 INFO pmon#supervisord: pcied   File "/usr/lib/python3/dist-packages/arista/libs/wait.py", line 53, in waitFor
2025 Jul 24 13:56:56.249217 CPH07-0101-0100-08T2-sup00 INFO pmon#supervisord: pcied     raise TimeoutError("Timed out waiting for %s" % description)
2025 Jul 24 13:56:56.249329 CPH07-0101-0100-08T2-sup00 INFO pmon#supervisord: pcied arista.libs.wait.TimeoutError: TimeoutError: Timed out waiting for eeprom sysfs entry (code 1)
2025 Jul 24 13:56:56.384451 CPH07-0101-0100-08T2-sup00 INFO pmon#supervisord 2025-07-24 13:56:56,383 WARN exited: pcied (exit status 1; not expected)
2025 Jul 24 13:56:57.386503 CPH07-0101-0100-08T2-sup00 INFO pmon#supervisord 2025-07-24 13:56:57,385 INFO spawned: 'pcied' with pid 222039
2025 Jul 24 13:57:07.397380 CPH07-0101-0100-08T2-sup00 INFO pmon#supervisord 2025-07-24 13:57:07,396 INFO success: pcied entered RUNNING state, process has stayed up for > than 10 seconds (startsecs)

wenyiz2021, 2025-07-29:
able to get eeprom on device:

WENYIZHANG@CPH07-0101-0100-08T2-sup00:~$ show platform syseeprom
ASY: ASY0522803D0
HwApi: 03.00
HwRev: 12.00
MAC: cc:1a:a3:ff:f4:20
MfgTime: 20200101000000
MfgTime2: 20240215191606
PCA: PCA0108806B0
SID: OtterLake
SKU: DCS-7800-SUP1A
SerialNumber: SGD240100C4

wenyiz2021, 2025-07-29:
the BT is timing out when checking this path existence:

 13    def eepromPath(self):
 14       return os.path.join(self.getSysfsPath(), 'eeprom')

wenyiz2021, 2025-07-29:
WENYIZHANG@CPH07-0101-0100-08T2-sup00:/sys/bus/pci/devices$ ls
0000:00:00.0  0000:00:1c.1  0000:05:00.0  0000:06:0b.0  0000:08:05.0  0000:12:03.0  0000:1c:01.0  0000:25:00.3  0000:2f:00.1  0000:33:00.0  0000:3a:0d.0  0000:51:00.2  0000:5f:00.0  0000:ff:0c.5  0000:ff:13.1  0000:ff:14.6
0000:00:01.0  0000:00:1c.2  0000:05:00.1  0000:06:0c.0  0000:08:08.0  0000:12:04.0  0000:1c:02.0  0000:25:00.4  0000:2f:00.2  0000:39:00.0  0000:3b:00.0  0000:51:00.3  0000:9f:00.0  0000:ff:0f.0  0000:ff:13.2  0000:ff:14.7
0000:00:01.1  0000:00:1d.0  0000:06:00.0  0000:06:0d.0  0000:08:0c.0  0000:12:05.0  0000:1c:03.0  0000:26:01.0  0000:2f:00.3  0000:39:00.1  0000:3d:00.0  0000:51:00.4  0000:a0:00.0  0000:ff:0f.4  0000:ff:13.3  0000:ff:15.0
0000:00:02.0  0000:00:1f.0  0000:06:01.0  0000:06:0e.0  0000:08:0d.0  0000:12:08.0  0000:1c:04.0  0000:26:02.0  0000:2f:00.4  0000:39:00.2  0000:46:00.0  0000:52:02.0  0000:a0:00.1  0000:ff:0f.5  0000:ff:13.4  0000:ff:15.1
0000:00:02.2  0000:00:1f.2  0000:06:02.0  0000:07:00.0  0000:09:00.0  0000:12:0c.0  0000:1c:05.0  0000:26:03.0  0000:30:01.0  0000:39:00.3  0000:46:00.1  0000:52:0d.0  0000:ff:0b.0  0000:ff:0f.6  0000:ff:13.5  0000:ff:15.2
0000:00:03.0  0000:00:1f.3  0000:06:03.0  0000:07:00.1  0000:0b:00.0  0000:12:0d.0  0000:1c:08.0  0000:26:04.0  0000:30:02.0  0000:39:00.4  0000:46:00.2  0000:54:00.0  0000:ff:0b.1  0000:ff:10.0  0000:ff:13.6  0000:ff:15.3
0000:00:05.0  0000:00:1f.6  0000:06:04.0  0000:07:00.2  0000:11:00.0  0000:13:00.0  0000:1c:0c.0  0000:26:05.0  0000:30:03.0  0000:3a:01.0  0000:46:00.3  0000:5c:00.0  0000:ff:0b.2  0000:ff:10.1  0000:ff:13.7  0000:ff:1e.0
0000:00:05.1  0000:03:00.0  0000:06:05.0  0000:07:00.3  0000:11:00.1  0000:15:00.0  0000:1c:0d.0  0000:26:08.0  0000:30:04.0  0000:3a:02.0  0000:46:00.4  0000:5c:00.1  0000:ff:0b.3  0000:ff:10.5  0000:ff:14.0  0000:ff:1e.1
0000:00:05.2  0000:03:00.1  0000:06:06.0  0000:07:00.4  0000:11:00.2  0000:1b:00.0  0000:1d:00.0  0000:26:0c.0  0000:30:05.0  0000:3a:03.0  0000:47:02.0  0000:5c:00.2  0000:ff:0c.0  0000:ff:10.6  0000:ff:14.1  0000:ff:1e.2
0000:00:05.4  0000:03:00.2  0000:06:07.0  0000:08:01.0  0000:11:00.3  0000:1b:00.1  0000:1f:00.0  0000:26:0d.0  0000:30:08.0  0000:3a:04.0  0000:47:0d.0  0000:5c:00.3  0000:ff:0c.1  0000:ff:10.7  0000:ff:14.2  0000:ff:1e.3
0000:00:14.0  0000:03:00.3  0000:06:08.0  0000:08:02.0  0000:11:00.4  0000:1b:00.2  0000:25:00.0  0000:27:00.0  0000:30:0c.0  0000:3a:05.0  0000:49:00.0  0000:5c:00.4  0000:ff:0c.2  0000:ff:12.0  0000:ff:14.3  0000:ff:1e.4
0000:00:1a.0  0000:04:00.0  0000:06:09.0  0000:08:03.0  0000:12:01.0  0000:1b:00.3  0000:25:00.1  0000:29:00.0  0000:30:0d.0  0000:3a:08.0  0000:51:00.0  0000:5d:02.0  0000:ff:0c.3  0000:ff:12.1  0000:ff:14.4  0000:ff:1f.0
0000:00:1c.0  0000:04:00.1  0000:06:0a.0  0000:08:04.0  0000:12:02.0  0000:1b:00.4  0000:25:00.2  0000:2f:00.0  0000:31:00.0  0000:3a:0c.0  0000:51:00.1  0000:5d:0d.0  0000:ff:0c.4  0000:ff:13.0  0000:ff:14.5  0000:ff:1f.2
WENYIZHANG@CPH07-0101-0100-08T2-sup00:/sys/bus/pci/devices$ cd /sys/devices
WENYIZHANG@CPH07-0101-0100-08T2-sup00:/sys/devices$ ls
breakpoint  cstate_core  intel_pt  LNXSYSTM:00  pci0000:00  platform  power     system      uncore_cbox_0  uncore_cbox_2  uncore_cbox_4  uncore_ha_0   uncore_imc_1  uncore_imc_3  uncore_r2pcie   uncore_r3qpi_1  uprobe
cpu         cstate_pkg   kprobe    msr          pci0000:ff  pnp0      software  tracepoint  uncore_cbox_1  uncore_cbox_3  uncore_cbox_5  uncore_imc_0  uncore_imc_2  uncore_pcu    uncore_r3qpi_0  uncore_ubox     virtual
WENYIZHANG@CPH07-0101-0100-08T2-sup00:/sys/devices$ ls eeprom

wenyiz2021, 2025-07-29:
@arista-nwolfe @Staphylo

Staphylo, 2025-07-30:
Hi @wenyiz2021 so apparently the failure happened when trying to identify the linecards from the supervisor.
In the backtrace you can see that `self.loadLinecards()` ends up calling `self.card.setupIdentification()` which initializes the necessary stuff to access the linecard eeprom and read it.
The call to `self.eeprom.setup()` will perform a `modprobe at24` followed by some write to the i2c bus `new_device` sysfs file to load the driver.
However, for some reason the device doesn't become reachable and there is a timeout.

The thing that would come to mind is a timing race where `pcied` is trying to do this before the platform is properly initialized.
Does this happen during boot or later after the device has been running for a while?
Do you see error messages in the `dmesg` that might be related?

wenyiz2021, 2025-07-30:
WENYIZHANG@CPH07-0101-0100-08T2-sup00:~$ ls -altr /sys/bus/i2c/devices/16-0050/
total 0
drwxr-xr-x 5 root root    0 Jul 18 03:38 ..
-rw-r--r-- 1 root root 4096 Jul 24 13:55 uevent
drwxr-xr-x 3 root root    0 Jul 24 13:55 .
lrwxrwxrwx 1 root root    0 Jul 24 13:55 subsystem -> ../../../../../../bus/i2c
-r--r--r-- 1 root root 4096 Jul 24 13:55 name
drwxr-xr-x 2 root root    0 Jul 30 18:10 power
-r--r--r-- 1 root root 4096 Jul 30 18:10 modalias
WENYIZHANG@CPH07-0101-0100-08T2-sup00:~$

wenyiz2021, 2025-07-30:
WENYIZHANG@CPH07-0101-0100-08T2-sup00: ls -latr /sys/bus/i2c/devices/16-0050/
total 0
drwxr-xr-x 5 root root    0 Jul 18 03:38 ..
-rw-r--r-- 1 root root 4096 Jul 24 13:55 uevent
drwxr-xr-x 3 root root    0 Jul 24 13:55 .
lrwxrwxrwx 1 root root    0 Jul 24 13:55 subsystem -> ../../../../../../bus/i2c
-r--r--r-- 1 root root 4096 Jul 24 13:55 name
drwxr-xr-x 2 root root    0 Jul 30 18:10 power
-r--r--r-- 1 root root 4096 Jul 30 18:10 modalias
WENYIZHANG@CPH07-0101-0100-08T2-sup00:~$ ls -latr /sys/bus/i2c/devices/15-0050/
ls: cannot access '/sys/bus/i2c/devices/15-0050/': No such file or directory
WENYIZHANG@CPH07-0101-0100-08T2-sup00:~$ ls -latr /sys/bus/i2c/devices/14-0050/
ls: cannot access '/sys/bus/i2c/devices/14-0050/': No such file or directory
WENYIZHANG@CPH07-0101-0100-08T2-sup00:~$ ls -latr /sys/bus/i2c/devices/13-0050/
ls: cannot access '/sys/bus/i2c/devices/13-0050/': No such file or directory
WENYIZHANG@CPH07-0101-0100-08T2-sup00:~$ ls -latr /sys/bus/i2c/devices/12-0050/
total 0
drwxr-xr-x 8 root root     0 Jul 18 03:38 ..
-rw-r--r-- 1 root root  4096 Jul 18 03:38 uevent
-rw------- 1 root root 65536 Jul 18 03:38 eeprom
drwxr-xr-x 4 root root     0 Jul 18 03:38 .
lrwxrwxrwx 1 root root     0 Jul 18 03:38 subsystem -> ../../../../../../bus/i2c
-r--r--r-- 1 root root  4096 Jul 18 03:38 name
drwxr-xr-x 3 root root     0 Jul 18 03:38 12-005014
lrwxrwxrwx 1 root root     0 Jul 30 19:19 supplier:regulator:regulator.0 -> ../../../../../virtual/devlink/regulator:regulator.0--i2c:12-0050
drwxr-xr-x 2 root root     0 Jul 30 19:19 power
-r--r--r-- 1 root root  4096 Jul 30 19:19 modalias
lrwxrwxrwx 1 root root     0 Jul 30 19:19 driver -> ../../../../../../bus/i2c/drivers/at24
WENYIZHANG@CPH07-0101-0100-08T2-sup00:~$ ls -latr /sys/bus/i2c/devices/11-0050/
total 0
drwxr-xr-x 8 root root     0 Jul 18 03:38 ..
-rw-r--r-- 1 root root  4096 Jul 18 03:38 uevent
-r--r--r-- 1 root root  4096 Jul 18 03:38 name
-rw------- 1 root root 65536 Jul 18 03:38 eeprom
drwxr-xr-x 4 root root     0 Jul 18 03:38 .
lrwxrwxrwx 1 root root     0 Jul 18 03:38 subsystem -> ../../../../../../bus/i2c
drwxr-xr-x 3 root root     0 Jul 18 03:38 11-005013
lrwxrwxrwx 1 root root     0 Jul 29 23:12 supplier:regulator:regulator.0 -> ../../../../../virtual/devlink/regulator:regulator.0--i2c:11-0050
drwxr-xr-x 2 root root     0 Jul 29 23:12 power
-r--r--r-- 1 root root  4096 Jul 29 23:12 modalias
lrwxrwxrwx 1 root root     0 Jul 29 23:12 driver -> ../../../../../../bus/i2c/drivers/at24
WENYIZHANG@CPH07-0101-0100-08T2-sup00:~$ ls -latr /sys/bus/i2c/devices/10-0050/
total 0
drwxr-xr-x 8 root root     0 Jul 18 03:38 ..
-rw-r--r-- 1 root root  4096 Jul 18 03:38 uevent
lrwxrwxrwx 1 root root     0 Jul 18 03:38 subsystem -> ../../../../../../bus/i2c
-rw------- 1 root root 65536 Jul 18 03:38 eeprom
drwxr-xr-x 4 root root     0 Jul 18 03:38 .
-r--r--r-- 1 root root  4096 Jul 18 03:38 name
drwxr-xr-x 3 root root     0 Jul 18 03:38 10-005012
lrwxrwxrwx 1 root root     0 Jul 30 19:20 supplier:regulator:regulator.0 -> ../../../../../virtual/devlink/regulator:regulator.0--i2c:10-0050
drwxr-xr-x 2 root root     0 Jul 30 19:20 power
-r--r--r-- 1 root root  4096 Jul 30 19:20 modalias
lrwxrwxrwx 1 root root     0 Jul 30 19:20 driver -> ../../../../../../bus/i2c/drivers/at24

wenyiz2021, 2025-07-30:
something happened on 7/24 that LC06 was plugged in

1 min before the stack:
2025 Jul 24 13:55:56.242441 CPH07-0101-0100-08T2-sup00 INFO kernel: [555478.605015] i2c i2c-16: new_device: Instantiated device 24c512 at 0x50

WENYIZHANG@CPH07-0101-0100-08T2-sup00:/sys/bus/i2c/devices$ cat /sys/bus/i2c/devices/16-0050/name
24c512

the directory /sys/bus/i2c/devices/16-0050 was created only on 7/24. whereas conversion completed on 7/18.

WENYIZHANG@CPH07-0101-0100-08T2-sup00:/sys/bus/i2c/devices$ ls -altr *0050
lrwxrwxrwx 1 root root 0 Jul 18 03:38 1-0050 -> ../../../devices/pci0000:ff/0000:ff:0b.3/i2c-1/1-0050
lrwxrwxrwx 1 root root 0 Jul 18 03:38 4-0050 -> ../../../devices/pci0000:ff/0000:ff:0b.3/i2c-4/4-0050
lrwxrwxrwx 1 root root 0 Jul 18 03:38 18-0050 -> ../../../devices/pci0000:00/0000:00:1c.0/0000:9f:00.0/i2c-18/18-0050
lrwxrwxrwx 1 root root 0 Jul 18 03:38 19-0050 -> ../../../devices/pci0000:00/0000:00:1c.0/0000:9f:00.0/i2c-19/19-0050
lrwxrwxrwx 1 root root 0 Jul 18 03:38 20-0050 -> ../../../devices/pci0000:00/0000:00:1c.0/0000:9f:00.0/i2c-20/20-0050
lrwxrwxrwx 1 root root 0 Jul 18 03:38 21-0050 -> ../../../devices/pci0000:00/0000:00:1c.0/0000:9f:00.0/i2c-21/21-0050
lrwxrwxrwx 1 root root 0 Jul 18 03:38 22-0050 -> ../../../devices/pci0000:00/0000:00:1c.0/0000:9f:00.0/i2c-22/22-0050
lrwxrwxrwx 1 root root 0 Jul 18 03:38 23-0050 -> ../../../devices/pci0000:00/0000:00:1c.0/0000:9f:00.0/i2c-23/23-0050
lrwxrwxrwx 1 root root 0 Jul 18 03:38 10-0050 -> ../../../devices/pci0000:00/0000:00:1c.0/0000:9f:00.0/i2c-10/10-0050
lrwxrwxrwx 1 root root 0 Jul 18 03:38 11-0050 -> ../../../devices/pci0000:00/0000:00:1c.0/0000:9f:00.0/i2c-11/11-0050
lrwxrwxrwx 1 root root 0 Jul 18 03:38 12-0050 -> ../../../devices/pci0000:00/0000:00:1c.0/0000:9f:00.0/i2c-12/12-0050
lrwxrwxrwx 1 root root 0 Jul 24 13:55 16-0050 -> ../../../devices/pci0000:00/0000:00:1c.0/0000:9f:00.0/i2c-16/16-0050
WENYIZHANG@CPH07-0101-0100-08T2-sup00:/sys/bus/i2c/devices$ cat /sys/bus/i2c/devices/16-0050/name
24c512

Thus, closing this issue as most likely some operation happened in the Data Center, will reopen unless we saw same issue again

wenyiz2021, 2025-08-02:
hi @arista-nwolfe , reopening this issue, we saw the same stack came back again.
Also we saw the LC slot 9 complaints on another T2, it might be image issue

wenyiz2021, 2025-08-04:
2025 Jul 30 23:19:54.210420 CPH07-0101-0100-08T2-sup00 WARNING kernel: [1107718.276978] scd 0000:9f:00.0: #0 cs { .reg=0x30002404, .rst=0 .fe=0, .ver=3, .brb=0, .nrq=0, .sp=0x0, .foe=1, .fsz=1, .nrs=4 } timed out after 108ms bus=16 addr=0x77
2025 Jul 30 23:19:54.210439 CPH07-0101-0100-08T2-sup00 WARNING kernel: [1107718.276978]  (scd_smbus_master_wait:266)

wenyiz2021, 2025-08-05:
/host: 921.4 MiB (966184960 bytes) trimmed on /dev/sda1
Watchdog armed for 180 seconds
Tue Aug 5 10:36:03 PM UTC 2025 Rebooting with platform x86_64-arista_7800_sup specific tool ...
INFO: Running powercycle script
INFO: Start graceful reboot on linecard WolverineQCpuMs(slotId=3)...
INFO: Start graceful reboot on linecard WolverineQCpuMs(slotId=4)...
INFO: Start graceful reboot on linecard WolverineQCpuMs(slotId=5)...
INFO: Graceful reboot on linecard WolverineQCpuMs(slotId=3) complete
INFO: Power off linecard WolverineQCpuMs(slotId=3) success
INFO: Graceful reboot on linecard WolverineQCpuMs(slotId=4) complete
ERROR: Power off linecard WolverineQCpuMs(slotId=4) failed
Traceback (most recent call last):
  File "/usr/lib/python3/dist-packages/arista/core/reboot.py", line 57, in _async_powerOffLinecard
   linecard.powerOnIs(False)
  File "/usr/lib/python3/dist-packages/arista/components/denali/card.py", line 195, in powerOnIs
   waitFor(lambda: not self.getUpstreamPort().reachable())
  File "/usr/lib/python3/dist-packages/arista/libs/wait.py", line 53, in waitFor
   raise TimeoutError("Timed out waiting for %s" % description)
arista.libs.wait.TimeoutError: TimeoutError: Timed out waiting for <lambda> (code 1)
WARNING: Linecard WolverineQCpuMs(slotId=5) shutdown timed out, forcing power off...
ERROR: Power off linecard WolverineQCpuMs(slotId=5) failed
Traceback (most recent call last):
  File "/usr/lib/python3/dist-packages/arista/core/reboot.py", line 57, in _async_powerOffLinecard
   linecard.powerOnIs(False)
  File "/usr/lib/python3/dist-packages/arista/components/denali/card.py", line 195, in powerOnIs
   waitFor(lambda: not self.getUpstreamPort().reachable())
  File "/usr/lib/python3/dist-packages/arista/libs/wait.py", line 53, in waitFor
   raise TimeoutError("Timed out waiting for %s" % description)
arista.libs.wait.TimeoutError: TimeoutError: Timed out waiting for <lambda> (code 1)
INFO: Initiating powercycle through SCD
INFO: Powercycle triggered by SCD

kenneth-arista, 2025-08-14:
As discussed offline, let's continue to monitor. Closing issue for now.